impatrq/silla_cdp #29: ATMega UART devuelve datos parcialmente.

GianK128, 2021-11-16:
En Proteus funciona correctamente el programa.

En la vida real devuelve datos a veces si y a veces no, pero los devuelve correctamente cuando lo hace.

GianK128, 2021-11-16:
Para probar:
- Testear UART con el ATMega puesto en el Arduino.
- Testear cristal para ver si oscila a 16MHz de verdad.

GianK128, 2021-11-16:
Ya probado:
- Varios cambios del Low Fuse para ver si era problema de la frecuencia. Al final quedo configurado con los fuses por defecto del Arduino.

GianK128, 2021-11-20:
Como solucion temporaria:

https://github.com/GianK128/silla_cdp/commit/e796725e9898d3bc62da6b7334e7bd470574e90b

GianK128, 2021-11-30:
Encontramos el problema:

Este estaba sucediendo debido a que la ESP32, al inicializar el puerto UART, envía una serie de bytes a través del TX, esto es lo que se consigue al imprimir vía consola lo que se le envía a un programa de eco. Estamos enviando la palabra `dummmy` (debe ser longitud 6 bytes):

```
Dummy write 0
b'\x04\x01\x10dum\x00'
Dummy write 1
b'mmydum\x00'
```

Lo que aparece en `Dummy write 0` es siempre una variación de diferentes caracteres, a veces mezclado con la palabra que le envío yo.
Y en `Dummy write 1` se puede ver el desfase que es provocado por la primera lectura problemática.

La solución que propongo entonces, es:
- Contar la cantidad de caracteres que está desfasada la lectura.
- Enviar tantos caracteres como desfase exista, para corregirla.

Todo esto antes de comenzar a ejecutar el programa principal.

Facunditenn, 2021-11-30:
Agregué el proceso de corrección de UART y lo implementé en el programa, ver #37.

GianK128, 2021-11-30:
No sé por qué no se cierran automáticamente.

closed 2021-11-30 by GianK128