impatrq/silla_cdp #36: ATMega problemas con ADC y lectura de joystick

GianK128, 2021-11-28:
- El ATMega al devolver datos devuelve valores de ADC entre 700 y 1023, nunca menor como deberia.
- El pulsador del joystick se esta leyendo siempre en 1 incluso teniendo una PullUp, asi que configurar en la programacion que lleva pullup
- Revisar tambien lo que se pueda de #29.

Facunditenn, 2021-11-28:
- Ya puse el PB4 como pull up en el init de puertos.
- En cuanto a #29, voy a seguir el consejo de Medina y probar con valores más altos de baudrates (probar con 76800, es el mas alto que no presenta error >2%)
[image]

Facunditenn, 2021-11-28:
Cuando antes pensabamos que no estaba respondiendo, resulta que si procesaba pero no habia respuesta para cuando no recibia o un "ASK" o un "MUX". Ahora cuando no entra a esos condicionales retorna "VALERR" (Value Error).

Gracias a esto podemos ver que:
- El problema no esta relacionado al baudrate porque persiste sin importar el baudrate.
- El problema se debe a que el atmega no logra entrar en los condicionales, por lo tanto o no esta recibiendo correctamente, o la ESP21 no está enviando correctamente.

GianK128, 2021-11-29:
Bueno, mañana le preguntaré a Medina si tiene algún consejo, y también voy a ver si puedo probar con otro ATMega u otra ESP32.

GianK128, 2021-11-30:
Ver [este comentario](https://github.com/impatrq/silla_cdp/issues/29#issuecomment-982235851) para desarrollo en #29.

Facunditenn, 2021-11-30:
Agregué el proceso de corrección de UART y lo implementé en el programa, ver #37.

GianK128, 2021-11-30:
No sé por qué no se cierran automáticamente.

Creo que este tambien debería estar.

closed 2021-11-30 by GianK128